TheImagingSource/IC3.5-Imaging-Control-Samples #79: Exposure Mode to Trigger Width with Device Trigger enabled using 3.4 SDK

smitanv, 2024-08-08:
How do I set the Exposure Mode to Trigger Width with Device Trigger enabled using 3.4 SDK? Thanks!

TIS-Stefan, 2024-08-08:
Hello

You set the Exposure Mode property element of Trigger accordingly to your needs. 

I am aware, this is a very short answer. I suppose, you want to do that programmatically, but you did not mention your the programming language you use. Therefore, I do not post source code. 

Also there is a small too, which creates the source code for all properties of a camera at https://github.com/TheImagingSource/IC-Imaging-Control-Samples/tree/master/Create%20Camera%20Properties%20Source%20Code. The [CreatePropertiesForCameras.zip](https://github.com/TheImagingSource/IC-Imaging-Control-Samples/blob/master/Create%20Camera%20Properties%20Source%20Code/CreatePropertiesForCameras.zip) should create the codes for IC 3.4. It creates files for C++, C# and VB.NET.

Stefan

smitanv, 2024-08-14:
Thanks Stefan. I used this code and it helps set the imaging control in trigger mode and accept physical triggers. I also set the exposure mode to "Trigger Width".
--------------------------------------------------------------------------------------------------------------------------------------
_imagingControl.DeviceTrigger = isExternalTriggerMode;
VCDPropertyItem triggerProp = _imagingControl.VCDPropertyItems.FindItem(VCDIDs.VCDID_TriggerMode);

VCDPropertyElement exposureRef = (VCDPropertyElement)triggerProp.Elements.FindElement("Exposure Mode");
VCDMapStringsProperty exposureMode = (VCDMapStringsProperty)exposureRef.FindInterface(VCDIDs.VCDInterface_MapStrings);
if (_imagingControl.DeviceTrigger)
{
    exposureMode.String = "Trigger Width";
}
else
{
    exposureMode.String = "Timed";
}
---------------------------------------------------------------------------------------------------------------------------------------
I have a couple related questions. When I have the above setting I notice lots of dropped frames when my frame rate is low at 30 fps. When I set frame rate to 60 fps or 120 fps I see better results. The physical external trigger width is set to 20 milliseconds.
1) Why does the frame rate need to be high to acquire the triggered frames and not miss frames?
2) What is meaning of "image readout"? I saw this line in one of the documents associated with the camera, DMK37BUX273, we are using -  "The duration of the image readout is the reciprocal of the current frame rate."

Thanks,
Smita

TIS-Stefan, 2024-08-15:
Hello Smita
I am happy to read, you got the properties set.
The frame rate in the camera determines the speed the camera runs with. If the frame rate is set to 30 fps, then the maximum speed of frame delivery is 30 frames, not more. if you trigger with 50Hz (50 frames per second) the camera does not speed up. 
Therefore, you need higher frame rate. Also please keep in mind: In trigger mode, the exposure time and the frame delivery time sum up. The frame delivery time is 1/fps. Thus, even if you use a trigger frequency of 30Hz and 30 frames per  second, you will encounter frame drops. 
You did it correctly by setting a higher frame rate. 
The "image readout" is reading the image data from the sensor and send it to the computer. It is, what I called "frame delivery time". 

We use the frame rate to reduce the amount of data transferred per second, so you can connect more than one camera to one connection controller. 

Stefan

smitanv, 2024-08-15:
Thanks Stefan. This is very helpful.

smitanv, 2024-08-16:
Hi Stefan,

I have one more question around external trigger and image buffer. There are the properties SampleStartTime and SampleEndTime on the ImageBuffer. What do they indicate? They do not seem to reflect the actual exposure time when the external trigger is on and the exposure mode is set to "Trigger Width". When I compute the difference between SampleStartTime for two back to back images, the difference is closer to how often the trigger is hit. 

Also, when I have IMX low-latency mode set to enabled, would that have any effect on the StartSampleTime? Is this value show when the trigger came in; or does it show when the exposure started for the frame?

Thanks,
Smita

TIS-Stefan, 2024-08-16:
The SampleStartTime is the computer's CPU time, when the driver was notified by the USB controller about an new frame has been completed. Please look at the Metadata at https://www.theimagingsource.com/en-us/documentation/icimagingcontrolcpp/c_structrefstruct_FrameMetaData.htm. The end time is simply calculated by adding frame rate. These times to not reflect the exposure time in any way.

The IMX Low Latency has an effect, because the the latency jitter between sending the trigger signal and exposure start is removed. You may look into the Technical Reference Manual of your camera on our website for a short description: 
If true:
Delay between trigger input and start of exposure is exactly
as configured through the Trigger Delay feature
else
Delay between trigger input and start of exposure is the
time configured by the Trigger Delay feature in addition to
a random delay depending on resolution, frame rate, and
timing relative to the sensor's internal state

Stefan

smitanv, 2024-08-19:
Thanks Stefan. I have a related question, does this mean that the device_time_stamp property on the MediaSampleInfo can be different from the SampleStartTime?

Thanks,
Smita

TIS-Stefan, 2024-08-20:
Hello
Yes. The drive and the device have different times. Also the camera has its own clock, while the driver uses the CPU clock. Therefore, the meta data contain the device time stamp and the driver time stamp. Same goes for the frame numbers. 
Stefan