TheImagingSource/IC3.5-Imaging-Control-Samples #81: Exposure time range is returned in different units for different camera types

PB-Ole, 2024-09-19:
Hi,

I hope someone can help me with this.

We have different camera types in use (DFK 33GX236 and DFK 37AUX290 for example) and are using the `tisgrabber` library to control the camera settings in our C++ application.
We want to allow the user to change the exposure time within a range limited a) by the camera itself and b) we set an upper limit of 33 ms to allow for a minimum of 30 FPS.
We use a call to
```
IC_CameraPropertyGetRange(grabber, CAMERA_PROPERTY::PROP_CAM_EXPOSURE,  &camExposureRangeMin, &camExposureRangeMax)
```
to obtain the min and max exposure time of the camera device. Now the problem is that for the DFK 37AUX290 this gives us
```
// unit not clear? 2^-15s to 2^4s?
camExposureRangeMin=-15
camExposureRangeMax=4
```
and for DFK 33GX236 we get
```
// unit is micro seconds
camExposureRangeMin=20
camExposureRangeMax=30'000'000
```
Is there a possibility to obtain the range in the same unit independent on the device? Or a way to know which unit is being returned by `IC_CameraPropertyGetRange`?

I have found the `IC_GetExpAbsValRange(grabber, fMin, fMax)` method which gives me the exposure range in micro seconds for the DFK 37AUX290 type, but for the DFK 33GX236 the application is crashing, even though (or because?) I check `IC_IsExpAbsValAvailable(grabber)`.

I would like to avoid hard-coded camera types and different handlings depending on them..

TIS-Stefan, 2024-09-19:
Hello

The -15 to +4 value range is the old style Physical Values interface. Do you have the correct driver version 5.1 installed for your camera?

Also I suggest to use the Absolute Values interface of exposure instead of range. Then you get the exposure times in fraction of seconds from both cameras. 

Additionally, I suggest to switch to IC Imaging Control 4 and discontinue using tisgrabber, if possible.

Stefan

PB-Ole, 2024-09-19:
Hi @TIS-Stefan 

many thanks for your very quick reply. Switching to IC Imaging Control 4 is at the moment unfortunately not possible for us. We do have the 5.1 driver version installed.

I would like to switch to the absolute values interface, but then I get problems with floating point precision.

If I do
```
float fMin, fMax;
IC_GetExpAbsValRange(grabber, &fMin, &fMax);
```
Then fMin is supposed to be 2e-5, as the minimum exposure duration for that camera model is 20 us. But this is not representable by a float, so it is in fact something like 1.999999995e-5.
So if I subsequently call
```
IC_SetExpAbsVal(grabber, fMin)
```
My application crashes, as `fMin` is outside the allowed range. I am already surprised that it crashes and does not simply ignore the invalid value, but that is probably another issue.

I will try to use double precision, but I am not sure how the tisgrabber library will handle this. Or do you have another suggestion what I should do?

Cheers,
Ole

TIS-Stefan, 2024-09-19:
Hello Ole
"1.999999995e-5." is a very nice float number, but it is displayed in the exponential way.
What does "crash" mean? Is there an error message?

PB-Ole, 2024-09-19:
It is a very nice float number, I agree, but tisgrabber does not seem to handle it well :)

There is no error message displayed. I have gdb attached and get until the line
```
IC_SetExpAbsVal(grabber, fMin)
```
where `fMin` is a float which was set by the preceeding call to `IC_GetExpAbsValRange(grabber, &fMin, &fMax);` and not changed by me after that. And there the crash happens. I see that fMin is 1.99...e-5 before the call and that I believe is the problem, because it is less than the allowed minimum value. Or am I missing something?

My current workaround is simply to add 1 mico second to `fMin`, since I don't really care about 1 us and then the application seems to be stable.

TIS-Stefan, 2024-09-19:
Hi Ole

Which compiler do you use?

Stefan

PB-Ole, 2024-09-19:
Hi Stefan,
we use MinGW with GCC 12.2 to compile our application on Windows 10. 
Cheers,
Ole

TIS-Stefan, 2024-09-20:
Hi Ole
I wonder, why your application ends without error message, when setting a value out of range. The function in the tisgrabber.dll checks for the range:

```C++
	/////////////////////////////////////////////////////////////////////////////////
	// Set the current value of a given property and element combination. 
	// AbsoluteValue interface is used
	// Property and Element names are not case sensitive
	int setPropertyAbsoluteValue(DShowLib::Grabber &Grabber, char* Property, char *Element, double Value)
	{
		int iSuccess = IC_SUCCESS;
		tIVCDAbsoluteValuePropertyPtr pAbsval = NULL;

		iSuccess = GetAbsValPropertyPtr(Grabber, Property, Element, pAbsval);

		if (iSuccess == IC_SUCCESS)
		{
			if (pAbsval != NULL)
			{
				if (pAbsval->getRangeMin() <= Value && Value <= pAbsval->getRangeMax())
					pAbsval->setValue(Value);
				else
					iSuccess = IC_INDEX_OUT_OF_RANGE; 
			}
		}
		return iSuccess;
	}
```
Maybe there is an error message in the the Windows EventViewer.
However, if it is working with your workaround, you should stay with that. 

Stefan