TheImagingSource/IC3.5-Imaging-Control-Samples #82: Simple question

SwHaraday, 2024-11-01:
Hi, wie geht's ?

Do you know what is going to happen, if you do below ?

ic.IC_SetFrameRate(hGrabber, ctypes.c_int(80))
* use 'ctypes.c_int' instead of 'ctypes.c_float'

We use DFK37BUX287.

Also please let ua know where to get latest 'tisgrabber_x64.dll' and related files.

Thanks in advance.

TIS-Stefan, 2024-11-01:
Hello

IC_SetFrameRate is declared as `int IC_SetFrameRate(HGRABBER hGrabber,float FrameRate);`
The IC_SetFrameRate function is not extra declared in tisgrabber.py. Thus I wonder where this change from float to int comes from. 

The latest version of the DLLs are available at https://github.com/TheImagingSource/IC-Imaging-Control-Samples/tree/master/Python/tisgrabber, which I am sure, you are aware of. 
We released IC Imaging Control 4 with a much better Python support and documentation for Windows and Linux, therefore, the tisgrabber.dll is end of development. 

Best regards
Stefan

SwHaraday, 2024-11-04:
Hi Stefan,

Thank you for your quick reply.

It's just from my strong assumtions.
While working on function to make different settings for each camera available, I made this stupid mistake.
However as far as my observation, it works 50% and does not work 50%.
Curious. That's why I wanted to know what is going happen. 

Minimal set of what we do is here.
[https://github.com/SwHaraday/multiple-camera-loader-for-YOLOv5](https://github.com/SwHaraday/multiple-camera-loader-for-YOLOv5)

By the way, where to learn 'IC Imaging Control 4' ?

Best Regards,
  Yuji

TIS-Stefan, 2024-11-04:
Hello Yuji

You can load and save presets with tisgrabber's IC_SaveDeviceStatetoFile() and IC_LoadDeviceStatefomFile functions.

IC 4:
To install the IC Imaging Control 4 Python Library into your active environment, run:
```
$ python3 -m pip install imagingcontrol4
```
If you want to use the pre-built dialogs for PySide6, you also have to install the imagingcontrol4pyside6 support library:
```
$ python3 -m pip install imagingcontrol4pyside6
```
Documentation is at https://www.theimagingsource.com/en-us/documentation/ic4python/guide-getting-started.html, samples are at https://github.com/TheImagingSource/ic4-examples/tree/master/python

It is not too different from tisgrabber, but it is better documentated, Linux compatible and uses the GenTL Producers in Windows and Linux instead of DirectShow / Aravis / v4l2. The GenTL Producers Windows are at https://www.theimagingsource.com/en-us/support/download/

Stefan

SwHaraday, 2024-11-04:
Hi Stefan,

Thank you for detailed information.
I will try.

Best Regards,
Yuji