TheImagingSource/IC3.5-Imaging-Control-Samples #83: Translate device_time_stamp

Robert42, 2025-02-13:
Hello,

we are using IC 3.5. In our use-case we are using a single camera and want to know when the picture  was taken.

The question is now how to use the [device_time_stamp](https://www.theimagingsource.com/en-us/documentation/icimagingcontrolcpp/prop_descMediaSampleInfo_device_time_stamp.htm). As already answered in #32, there is no explicit way of getting the current device time.
So the only way to get the current time is to take an image at a known time and look at the frame metadata what device_time_stamp the known time corresponds to.

I have a few questions about the best practice to do that:

1. Is  there a significant clock drift in the device_time_stamp?

     For example QueryPerformanceCounter has according to [the documentation](https://learn.microsoft.com/en-us/windows/win32/sysinfo/acquiring-high-resolution-time-stamps) typically a clock drift of 30-50 parts per million (and rarely up to 500 ppm).

    Am I right to assume that the device_time_stamp has a clock drift that's <= the clock drift of QueryPerformanceCounter?

2. How can I synchronize a clock on the computer with the device clock?

    1. One option is to enable the trigger mode, call QueryPerformanceCounter and snap one picture, add the trigger delay (`VCDElement_TriggerDelay` after converting the unit) to the QPC value.

        And now I have a known QPC/device_time_stamp pair

    2. When I call `prepareLive` and then `startLive`, is the call to `startLive` the trigger of the first image? In other words: Can I call `prepareLive`, then call`QueryPerformanceCounter` rigth before `startLive` (and add the delay) to know when the exposure of first image started?

        In order to address clock drift, I would need to synchronize after an amount of time. Could I simply `suspendLive`, call QueryPerformanceCounter and `startLive` to get another matching QPC/device_time_stamp pair?

3. Is there any other way of getting the unit of the device time stamp other than comparing the device_time_stamp values of two images with known fps?

TIS-Stefan, 2025-02-13:
Hello 
The clock in  the camera does not drift significantly. But we never measured that, thus, we can not qualify it. 

The clock can not be synchronized. If you have a GigE camera, the clock of the camera can be reset to 0 using action commands. It is more easy to use Action Commands with IC Imaging Control 4 instead of 3.5. In case you have a PTP device in your computer and use GigE cameras, the cameras could synchronize with that PTP device automatically. 

Suspend Live and Startlive do not reset the clock in the camera. The clock in the camera starts at 0, when the camera is powered. 
The unit of time stamps is 100ns ticks, if I am right. 

Stefan

TIS-Stefan, 2025-02-13:
Hello

Which camera model do you use?

Stefan

Robert42, 2025-02-14:
Hello Stefan,

we use the models DMK 37BUX287 and DMK 37AUX287

Robert

TIS-Stefan, 2025-02-14:
Hello Robert

The cameras have a clock with +-50 ppm.

suspendLive and startlive do not reset the clocks of the USB cameras.

The unit of the time stamp is nano seconds. 

Your option 2i is, what I would do too for a synchronization.

In case you need an exact synchronisation, you may consider using network GigE cameras and a PtP Time Server. 

Stefan

Robert42, 2025-02-14:
Awesome! Thank you for the answers!

closed 2025-02-14 by Robert42