TheImagingSource/IC3.5-Imaging-Control-Samples #84: Can I use a webcam with IC Imaging Control 3.5?

GLC-Kazetani, 2025-03-06:
Hello everyone.

I am trying to control THE IMAGING SOURCE camera and a webcam interchangeably using IC Imaging Control 3.5. 
I have connected both devices, but the webcam does not seem to be recognized. 
Does IC Imaging Control 3.5 support webcams?

Thank you for your assistance.

TIS-Kevin, 2025-03-06:
Hello,

"Does IC Imaging Control 3.5 support webcams?"

No we do not support webcams.
IC Imaging Control does not support Cameras from any other Producers.

Best Regards,

Kevin

GLC-Kazetani, 2025-03-06:
Hello, Kevin

Thank you for getting back to me. 
I now understand that IC Imaging Control 3.5 does not support webcams. 

Thank you for the information.

closed 2025-03-06 by GLC-Kazetani